lacymorrow/fpv-bible #3: Micro H Brushed Review

lacymorrow, 2016-10-22:
Spoiler: Shit

lazd, 2016-10-22:
Yeah almost not worth it. Maybe something more trendy, like we take an review an Eachine inductix clone, note that it's OK but kinda shit, then upgrade it with a Beebrain, retrofit it to use normal Eflite batteries, and build it Shiny Whoop style.

lacymorrow, 2016-10-23:
Honestly our target demographic is pretty wide, i'm kind of of the opinion that reviewing anything and everything, crap included, is something that will set our content apart.

It's a pretty logical "next step" in FPV to people who don't know better, me and you both made the same mistake, I think we could make it into a pretty interesting conversation about why we thought it was good at the time and why it actually turned out to be shitty.

The more "real" the experience, the better. I imagine our mission statement to be something along the lines of: *"Providing the true, full FPV experience. Vicariously."*

Maybe I'm wrong but I think noobs who want to get in but never will (financial reasons or other) are going to be our primary viewers. The accuracy and personal experience from us will eventually give us a solid rep.

lacymorrow, 2016-10-23:
Dunno if I'm right, but I think if we can make the format interesting we should do all sorts of topics, not just review the good shit.